koquantam/android_device_samsung_vivalto3gvn #1: Cannot run logcat

ngoquang2708, 2016-01-06:
adbd is running so I can use adb shell to get into the phone, but neither adb logcat nor logcat in adb shell work, it just hang there with no output.

ngoquang2708, 2016-01-06:
Sorry. It work now. I have no idea why!

closed 2016-01-06 by ngoquang2708

koquantam, 2016-01-06:
How???? I encountered this error too.
The only thing i could do when this phone in CM11 is adb shell. No idea why adb logcat doesn't work (showing blinking cursor)

ngoquang2708, 2016-01-06:
It does not work with that prebuilt kernel. You have to enable the kernel build config to build from source. But there is a problem in the mali gpu driver source. I am fixing it. Here my config:

# kernel
BOARD_KERNEL_BASE := 0x00000000
BOARD_KERNEL_CMDLINE := console=ttyS1,115200n8
BOARD_KERNEL_PAGESIZE := 2048
TARGET_KERNEL_CONFIG := vivalto3g-native_hw04_defconfig
TARGET_KERNEL_SOURCE := kernel/samsung/vivalto3gvn
#TARGET_PREBUILT_KERNEL := device/samsung/vivalto3gvn/prebuilt/kernel

Remember to setup the kernel source code correctly before building.

koquantam, 2016-01-06:
So does it boots?

ngoquang2708, 2016-01-06:
No. There is some problems in the proprietary-files.txt that cause surfaceflinger to crash. You have to remove those libraries

lib/libEGL.so
lib/libETC1.so
lib/libGLES_trace.so
lib/libGLESv1_CM.so
lib/libGLESv2.so
lib/libGLESv3.so
lib/libOpenMAXAL.so
lib/libOpenSLES.so

koquantam, 2016-01-06:
Okay, please add pull requests to me, i'll handle it

ngoquang2708, 2016-01-06:
OK