koquantam/android_device_samsung_vivalto3gvn #10: Video decoding

koquantam, 2016-01-08:
Chrome FC when going to play a video on YouTube

ngoquang2708, 2016-01-12:
VLC can play mp4 files but the build in player is not. So does YouTube too.

ngoquang2708, 2016-01-21:
When using hw decoding, I get this:

I/SprdOMXPlugin(  129): makeComponentInstance 'OMX.sprd.h264.decoder'
I/SPRDAVCDecoder(  129): Construct SPRDAVCDecoder, this: b85a7110
I/SPRDAVCDecoder(  129): SPRDAVCDecoder, video_cfg: 0
I/SPRDAVCDecoder(  129): openDecoder, lib: libomx_avcdec_hw_sprd.so
I/SPRDAVCDecoder(  129): SPRDAVCDecoder, is IOMMU enabled: 1
I/SPRDAVCDecoder(  129): pmem 0x20000000 - 0xb29cb000 - 2097152
D/OMXCodec(  129): Successfully allocated OMX node 'OMX.sprd.h264.decoder'
I/OMXCodec(  129): [OMX.sprd.h264.decoder] AVC profile = 66 (Baseline), level = 30
I/SPRDAVCDecoder(  129): internalGetParameter, OMX_IndexParamVideoPortFormat, eColorFormat: 0x15
E/SPRDAVCDecoder(  129): Failed to open /sys/devices/platform/scxx30-dmcfreq.0/devfreq/scxx30-dmcfreq.0/ondemand/set_freq
I/OMXCodec(  129): [OMX.sprd.h264.decoder] video dimensions are 640 x 360
I/OMXCodec(  129): [OMX.sprd.h264.decoder] Crop rect is 640 x 360 @ (0, 0)
I/SPRDAVCDecoder(  129): internalSetParameter, enable AndroidNativeBuffer
I/OMXCodec(  129): [OMX.sprd.h264.decoder] allocating 8 buffers of size 131072 on input port
I/OMXCodec(  129): [OMX.sprd.h264.decoder] allocated buffer 0xb85c01b8 on input port
I/OMXCodec(  129): [OMX.sprd.h264.decoder] allocated buffer 0xb85c0380 on input port
I/OMXCodec(  129): [OMX.sprd.h264.decoder] allocated buffer 0xb85c0450 on input port
I/OMXCodec(  129): [OMX.sprd.h264.decoder] allocated buffer 0xb85c0520 on input port
I/OMXCodec(  129): [OMX.sprd.h264.decoder] allocated buffer 0xb85c05f0 on input port
I/OMXCodec(  129): [OMX.sprd.h264.decoder] allocated buffer 0xb85c0288 on input port
I/OMXCodec(  129): [OMX.sprd.h264.decoder] allocated buffer 0xb85c0850 on input port
I/OMXCodec(  129): [OMX.sprd.h264.decoder] allocated buffer 0xb85c0908 on input port
I/SPRDAVCDecoder(  129): internalGetParameter, OMX_IndexParamGetAndroidNativeBuffer 33
W/GraphicBufferAllocator(  125): alloc(640, 360, 21, 00002933, ...) failed -22 (Invalid argument)
E/        (  125): GraphicBufferAlloc::createGraphicBuffer(w=640, h=360) failed (Invalid argument), handle=0x0
E/BufferQueue(  125): [SurfaceView] dequeueBuffer: SurfaceComposer::createGraphicBuffer failed
E/OMXCodec(  129): dequeueBuffer failed: Invalid argument (22)
E/OMXCodec(  129): [OMX.sprd.h264.decoder] Allocate Buffer failed - error = -22
E/OMXCodec(  129): [OMX.sprd.h264.decoder] init failed: -22
I/OMXNodeInstance(  129): OMX_FreeBuffer for buffer header 0xb85c0908 successful
I/OMXNodeInstance(  129): OMX_FreeBuffer for buffer header 0xb85c0850 successful
I/OMXNodeInstance(  129): OMX_FreeBuffer for buffer header 0xb85c0288 successful
I/OMXNodeInstance(  129): OMX_FreeBuffer for buffer header 0xb85c05f0 successful
I/OMXNodeInstance(  129): OMX_FreeBuffer for buffer header 0xb85c0520 successful
I/OMXNodeInstance(  129): OMX_FreeBuffer for buffer header 0xb85c0450 successful
I/OMXNodeInstance(  129): OMX_FreeBuffer for buffer header 0xb85c0380 successful
I/OMXNodeInstance(  129): OMX_FreeBuffer for buffer header 0xb85c01b8 successful
E/AwesomePlayer(  129): failed to start video source
I/SPRDAVCDecoder(  129): Destruct SPRDAVCDecoder, this: b85a7110
E/SPRDAVCDecoder(  129): Failed to open /sys/devices/platform/scxx30-dmcfreq.0/devfreq/scxx30-dmcfreq.0/ondemand/set_freq
I/ActivityManager(  543): Displayed com.android.gallery3d/.app.MovieActivity: +1s361ms
I/Timeline(  543): Timeline: Activity_windows_visible id: ActivityRecord{41b6b170 u0 com.android.gallery3d/.app.MovieActivity t4} time:313584
D/LIGHTS  (  543): file:device/samsung/vivalto3gvn/libs/liblights/lights.c, func:set_light_buttons, on=0
E/MediaPlayer( 1751): error (1, -22)
E/MediaPlayer( 1751): Error (1,-22)
D/CodeauroraVideoView( 1751): Error: 1,-22
I/SPRDHWComposer(  125): Do not support display type: 0
D/ENGPC   (  144): parse_event: event { 'change', '/devices/platform/sec-battery/power_supply/battery', 'power_supply', '' }
I/SPRDHWComposer(  125): Do not support display type: 0
I/SPRDHWComposer(  125): Do not support display type: 0
I/SPRDHWComposer(  125): Do not support display type: 0
I/SPRDHWComposer(  125): Do not support display type: 0
I/SPRDHWComposer(  125): Do not support display type: 0
I/SPRDHWComposer(  125): Do not support display type: 0

ngoquang2708, 2016-01-21:
It is failed at this step:
```
W/GraphicBufferAllocator(  125): alloc(640, 360, 21, 00002933, ...) failed -22 (Invalid argument)
```

koquantam, 2016-01-21:
Does video player FC? I got that too

koquantam, 2016-01-21:
Well, our stock is completely damn broken. It continuously spamming logcat, espically MTP, so i can't see anything much to compare

ngoquang2708, 2016-01-21:
Everything is working fine (but S/W decoding, of course) except video player.

ngoquang2708, 2016-01-21:
Some app have its display overflow the screen. Adding density prop = 240 fix that for AOSP Keyboard. But google voice search still got that bug.

koquantam, 2016-01-21:
Hmm... you tried gapps already? I haven't, because it's very laggy so nano gapps is my best choice
S/W decoding works, but it's damn slow (pictures showing slowish, like 8 yrs old bride :D), even MX Player can't doing better. Maybe Sammy has done sth to libstagefrighthw so its size is bigger than us

koquantam, 2016-01-21:
What..... mali.ko is over 225kb?
Why i got it over 4mb?

ngoquang2708, 2016-01-21:
Maybe that is a un-stripped version. Mine is stripped out debug symbols, because it is in kernel source tree so I think that CM does that automatically.

ngoquang2708, 2016-01-21:
You use nano, I use pico :)

ngoquang2708, 2016-01-21:
yeah, that true, our libstagefrighthw do not work as we intended. The true fact is, it does not work at all because it does not have ```createOMXPlugin``` function. I just copy it from stock for testing.

koquantam, 2016-01-21:
So that means you don't have mali in /sys/kernel/debug

ngoquang2708, 2016-01-21:
Just checked. It is there.

koquantam, 2016-01-21:
Did you just reserved-engineering the libstagefrighthw?

ngoquang2708, 2016-01-21:
I ```grep``` .so binary :D

koquantam, 2016-01-21:
Well that's a good trick

ngoquang2708, 2016-01-21:
Reserved-engineering is out of my knowledge :)

ngoquang2708, 2016-01-21:
I look at the code that load libstagefrighthw.so, and it manage to call that function, our does not have that, and SS blob does have that.

koquantam, 2016-01-23:
Hmm... I tried your recent build and this thing seems working fine, but it currently doesn't support 480p on YouTube

ngoquang2708, 2016-01-23:
What? On browser or on YT app?

koquantam, 2016-01-23:
YouTube app. It doesn't FC like before. Haven't tried video playback on FB yet

ngoquang2708, 2016-01-23:
I never get YouTube app to work.

koquantam, 2016-01-23:
Video playback on FB doesn't work. Only sounds, no image

ngoquang2708, 2016-01-23:
YouTube app can play low res video because it use S/W decoding, and S/W decoding have max res of 320/240. So, I need to use H/W decoding to play higher res video, and it fail. So H/W decoding is still not working now.

koquantam, 2016-01-24:
Yeah, i saw logcat. Needs to get more sources :D

koquantam, 2016-01-24:
media_codecs.xml?

koquantam, 2016-01-24:
There's a bunch of changes that we've missed from them. Check sprd_frameworks_av history

koquantam, 2016-01-26:
Our stock libstagefrighthw works fine, but somehow it got slowish (even in stock, it still slow like that, but will get fast immediately)

ngoquang2708, 2016-01-26:
Could it play 480p ot higher res?

koquantam, 2016-01-26:
In MX Player only, other apps FC

ngoquang2708, 2016-01-26:
I think mx is using its own s/w decoding, just like vlc does, not use h/w.

koquantam, 2016-01-26:
Oh, i'm wrong again :(

koquantam, 2016-01-26:
But, in VLC, when using full acceleration, it just acting slowish like that

koquantam, 2016-01-26:
In the other words: it lags

ngoquang2708, 2016-01-26:
That true. So, to confirm that whether h/w decoding is working or not, use
YouTube app.

koquantam, 2016-01-26:
In stock, it could locate the hw en/decoding libs, but in CM it doesn't

koquantam, 2016-01-30:
I've read somewhere that only 4.4.2 got 480p decoding. Gotta check it out and reports

ngoquang2708, 2016-01-30:
We are 4.4.4!??

koquantam, 2016-01-30:
Nah, they said just change android version to 4.4.2, LOL

ngoquang2708, 2016-01-30:
??? Where are those info?

koquantam, 2016-01-30:
https://forums.oneplus.net/threads/guide-how-to-enable-1080p-youtube-on-all-1080p-android-devices.86854/

ngoquang2708, 2016-01-30:
Did you test it?

koquantam, 2016-01-30:
Yeah, already tested and same results, no changes
I bet they just kidding around everyone :D

ngoquang2708, 2016-01-30:
:D

koquantam, 2016-01-31:
Hmm... why stock doesn't have this?
E/SPRDAVCDecoder(  129): Failed to open /sys/devices/platform/scxx30-dmcfreq.0/devfreq/scxx30-dmcfreq.0/ondemand/set_freq
Both compiled and stock kernel don't. Probably Sammy has done sth with it

ngoquang2708, 2016-01-31:
The phone is randomly dead with no reason, do you encountered it?

koquantam, 2016-01-31:
If with camera then yes, if it is soft-rebooted then IDK, probably system_server crashed

koquantam, 2016-01-31:
Now i can confirm that the phone just dead with no reason

ngoquang2708, 2016-01-31:
Maybe cpu freq in power_config.xml cause this, I make a update, compiling it.

koquantam, 2016-01-31:
cpu-active is too damn slow. Maybe that's the reason

ngoquang2708, 2016-01-31:
But that came from stock, sigh, stock again

koquantam, 2016-01-31:
It dead after pressing power button (fully slept?)

ngoquang2708, 2016-01-31:
cpu.active is used to mesure power comsumtion, cpu.speeds is the one that get interested

ngoquang2708, 2016-01-31:
Investigating...

koquantam, 2016-01-31:
You know what, stock has DVFS, which use to lock the CPU speeds itself, so they could kidding around the frameworks

ngoquang2708, 2016-01-31:
Likes the battery capacity :D

koquantam, 2016-01-31:
Yeah :D
Like, a core prime has 2000, but by some magic, it has increased to 2500, LOL

ngoquang2708, 2016-01-31:
Stock have this too, man:
```
E/SPRDAVCDecoder(  137): Failed to open /sys/devices/platform/scxx30-dmcfreq.0/devfreq/scxx30-dmcfreq.0/ondemand/set_freq
```

koquantam, 2016-01-31:
Well, good logcat digging
I'm still finding a way to fix this w/o stock blobs

[48 more replies]

closed 2016-02-06 by koquantam