koquantam/android_device_samsung_vivalto3gvn #11: No WiFi direct

ngoquang2708, 2016-01-08:


koquantam, 2016-01-08:
There's no wifidirect.apk in /system/app so no wifi direct

ngoquang2708, 2016-01-08:
Yes. Did note it here for later.

ngoquang2708, 2016-02-05:
I can pair wifi direct with other device, but I can not send anything because sento menu do not have wifi direct, maybe 3rd app can help.

koquantam, 2016-02-05:
Really? Which in Settings?

ngoquang2708, 2016-02-05:
Go to Wifi settings, touch [...] button

ngoquang2708, 2016-02-05:
And there you go, Wi-Fi Direct menu shows up

koquantam, 2016-02-05:
Nah, doesn't exist

ngoquang2708, 2016-02-05:
I added wifi direct xml with this commit 12a6dad2b82b5858fd064a3ac6b0347e1ab746f8

ngoquang2708, 2016-02-05:
Check your /system/etc/to/permissions/ to see if wifi.direct.xml is there, if not, copy it to there.

koquantam, 2016-02-05:
OK, i'll do it later

ngoquang2708, 2016-02-05:
Note: do not add ```<bool translatable="false" name="config_wifi_p2p_support">true</bool>``` to ```config.xml```, it is unsupported option, I removed with the later commit though.

ngoquang2708, 2016-02-06:
Please check this.

koquantam, 2016-02-06:
Confirmed
It appears now, but i don't have a device to check

closed 2016-02-06 by koquantam