koquantam/android_device_samsung_vivalto3gvn #12: GPS not work properly

ngoquang2708, 2016-01-08:


ngoquang2708, 2016-01-09:
GPS is now working probably with pull request #14 . So close this issue.

koquantam, 2016-01-10:
More works to get done :)
Now the only think we need to fix is RIL and bluetooth.

koquantam, 2016-02-05:
Hmm... GPS doesn't work at all
It just turns on, cLock (CM clock widget) can't get location

ngoquang2708, 2016-02-05:
IDK, I can use it, GMaps works with it.

koquantam, 2016-02-05:
Weird. Can you test it with cLock?

ngoquang2708, 2016-02-05:
It works for me now. Try remove it. Then re-add it, when reset it, it will show options to enable weather, chose source as OpenWeatherMap, then tick the accept mark, wait for a couple of minutes.

koquantam, 2016-02-05:
It doesn't work out

ngoquang2708, 2016-02-05:
Did I make some other derp actions to make it works, like audio rec :D

koquantam, 2016-02-05:
Well, i'm the one who got derp now LOL

koquantam, 2016-02-06:
Somehow it actually works now LOL

closed 2016-02-06 by koquantam