koquantam/android_device_samsung_vivalto3gvn #13: hwcomposer (maybe a fix in patches)

koquantam, 2016-01-08:
There's a hwcomposer in patches/frameworks/native
Should we try that patch?

ngoquang2708, 2016-01-08:
Let's try it to see what it does.

koquantam, 2016-01-08:
Make a backup first (to prevent repo sync again)

ngoquang2708, 2016-01-08:
Yeah

koquantam, 2016-01-08:
Any difference? I just got lost power recently

koquantam, 2016-01-08:
Any ideas about this/
E/HAL     (  121): load: module=/system/lib/hw/hwcomposer.scx15.so
E/HAL     (  121): dlopen failed: cannot locate symbol "_ZN7android13MemoryHeapIon21Get_phy_addr_from_ionEiPiS1_" referenced by "hwcomposer.scx15.so"...
E/SurfaceFlinger(  121): hwcomposer module not found

ngoquang2708, 2016-01-08:
Like I mentioned before, libbinder in frameworks/native/libs/binder lacks the symbol for hwc.scx15. The libbinder.so blob from stock rom does have that symbol. But when copy it to the phone, it make the phone unstable.

koquantam, 2016-01-08:
What do you mean "unstable"?

ngoquang2708, 2016-01-08:
It spam logcat whit unknown events and other things that I don't remember. You could experience it yourself.

koquantam, 2016-01-08:
like "Enable myanmar Zawgyi converter"?
Stock shows that a lot

ngoquang2708, 2016-01-08:
I don't remember clearly, some things like E/SPRDHwcomposer ....

koquantam, 2016-01-08:
Alright replaced stock libbinder.so and it shows a bunch of this:
I/Netd    (  287): Netd 1.0 starting

ngoquang2708, 2016-01-08:
I searched google and found no result of Get_phy_addr_from_ion. I think that is samsung private code.

koquantam, 2016-01-08:
LOL, a bunch of:
E/SPRDHWComposer( 439): unsupported event

ngoquang2708, 2016-01-08:
That it!

ngoquang2708, 2016-01-08:
MemoryHeapBase::getSize() may be this is our Get_phy_addr_from_ion ????

ngoquang2708, 2016-01-08:
or this uint32_t MemoryHeapBase::getOffset() ???

koquantam, 2016-01-08:
I'm searching for in from Spreadtrum sources

ngoquang2708, 2016-01-08:
OMG, C++. Why?

koquantam, 2016-01-08:
We need a patch to libbinder so get hwc working

ngoquang2708, 2016-01-08:
No HWC source code, that is a very hard work to do. My priority now is focused to RIL so at least I dont have to restore to stock rom to make a phone call!!!

koquantam, 2016-01-08:
Well at least i've found this:
https://git.mozilla.org/?pf=external/sprd-aosp
Maybe there's sth related to out device

ngoquang2708, 2016-01-08:
FirefoxOS is dead, and those sources is gone too, soo sad.

koquantam, 2016-01-08:
Oh yeah, just found sth
Anyway, mint got a libbinder fix and got hwc working. I'm asking for the dev to post his patches

koquantam, 2016-01-08:
Alright, we got the patches here, for libbinder (not a diff patch, just copy-paste to source)
https://github.com/TeamButter/patches?files=1
Gotta get some sleeps now :)

ngoquang2708, 2016-01-09:
Oh yeahhhhhhhhhhhhhhhhhhhh

koquantam, 2016-01-09:
Still doesn't work, shows a lot unsupported event in logcat

ngoquang2708, 2016-01-09:
I think it need some modification in the Android.mk makefile

$(TARGET_OUT_INTERMEDIATES)/KERNEL/

on my system is 

$(TARGET_OUT_INTERMEDIATES)/KERNEL_OBJ/

Yours?

ngoquang2708, 2016-01-09:
Please folk my android_frameworks_native, hardware_libhardware, hardware_libhardware_legacy to sync our works.

koquantam, 2016-01-09:
$(TARGET_OUT)INTERMEDIATES)/KERNEL_OBJ/

ngoquang2708, 2016-01-09:
in folder out/target/product/vivalto3gvn/obj. there is no KERNEL folder. but KERNEL_OBJ. and libbinder Android.mk use KERNel. please check

koquantam, 2016-01-09:
Oh yeah, that's real
Gotta change it to see any differences

koquantam, 2016-01-09:
Nothing changed, still get spamming logcat

ngoquang2708, 2016-01-09:
Does hwc complain about libbinder any more?

koquantam, 2016-01-09:
Like what i said, it keep spamming logcat

koquantam, 2016-01-09:
I got an idea
Try building these blobs with the ROM (not using vendor anymore) and test if it works. They're buildable, i've built this before

ngoquang2708, 2016-01-09:
but where do you get those sources?

koquantam, 2016-01-09:
IDK, just add it to device.mk, it will build.

ngoquang2708, 2016-01-09:
I see. It will not build anything and just ignore it.

ngoquang2708, 2016-01-09:
Please fork my system_core for libion fix. Then you can verify hwc is now working, and close this issue.

koquantam, 2016-01-10:
OK, will test it

ngoquang2708, 2016-01-10:
Oh man. You fork a wrong one. Sorry if confused you but the right one is android_system_core

ngoquang2708, 2016-01-10:
I fork android_system_core and fix it libion base on libion of sprd_system_core

koquantam, 2016-01-10:
LOL, still get spamming logcat
Could you tell me how to fix that again?

ngoquang2708, 2016-01-10:
<?xml version="1.0" encoding="UTF-8"?>
<manifest>

	<remove-project name="CyanogenMod/android_frameworks_native" />
	<project path="frameworks/native" name="koquantam/android_frameworks_native" remote="github" revision="cm-11.0"/>

	<remove-project name="CyanogenMod/android_hardware_libhardware" />
	<project path="hardware/libhardware" name="koquantam/android_hardware_libhardware" remote="github" revision="cm-11.0"/>

	<remove-project name="CyanogenMod/android_hardware_libhardware_legacy" />
	<project path="hardware/libhardware_legacy" name="koquantam/android_hardware_libhardware_legacy" remote="github" revision="cm-11.0"/>

	<remove-project name="CyanogenMod/android_system_core" />
	<project path="system/core" name="koquantam/android_system_core" remote="github" revision="cm-11.0"/>

</manifest>

ngoquang2708, 2016-01-10:
Download this https://gist.github.com/ngoquang2708/949ca992316c4b132b94
COpy and past it to your cm11-0/.repo/local_manifests/
Delete framewords/native, hardware/libhadrware, hardware/libharware_legacy, system/core
Then repo sync
THen compile

koquantam, 2016-01-10:
OK, will try

koquantam, 2016-01-10:
Got this:
https://pastee.org/xbe6j

ngoquang2708, 2016-01-10:
repo sync --force-sync hardware/libhardware

if faced it again, then force sync again, 4 times at most :D

koquantam, 2016-01-10:
I use repo sync --force-sync
It has no syntax behind, so use that once :)

ngoquang2708, 2016-01-10:
I copy that from your pastes though

koquantam, 2016-01-10:
So, now does it smooth? ex: swiping notification bar?

ngoquang2708, 2016-01-10:
I now use HWC so yes

koquantam, 2016-01-10:
GPS doesn't work properly on Quick Settings Bar (only have two options: Off and Device only)

koquantam, 2016-01-10:
Now let's focus on RIL. This ROM is almost good for daily use.

ngoquang2708, 2016-01-10:
Device only is GPS.

RIL is now work with just one SIM. But because of audio recoding problem, we cannot make a outgoing call. What make me confuse is that why we can receive a call?

koquantam, 2016-01-10:
It doesn't work, it shows this in logcat (and stuck on Samsung logo)
https://pastee.org/2mf4y

ngoquang2708, 2016-01-10:
Check your out/target/product/vivalto3gvn/obj/KERNEL_OBJ/usr/include/linux/ion.h file to see if it have ION_IOC_INVALIDATE definition

ngoquang2708, 2016-01-10:
Here my build so you can check out http://qrpi.noip.me:10080/upload/cm-11-20160110-UNOFFICIAL-vivalto3gvn.zip

koquantam, 2016-01-10:
it's defined

ngoquang2708, 2016-01-10:
Does system/core/libion/Android.mk look like this

LOCAL_PATH:= $(call my-dir)
 
 include $(CLEAR_VARS)
LOCAL_C_INCLUDES += \
    $(TARGET_OUT_INTERMEDIATES)/KERNEL_OBJ/usr/include/
 LOCAL_SRC_FILES := ion.c
 LOCAL_MODULE := libion
 LOCAL_MODULE_TAGS := optional
@@ -8,6 +10,8 @@ LOCAL_SHARED_LIBRARIES := liblog
 include $(BUILD_SHARED_LIBRARY)
 
 include $(CLEAR_VARS)
LOCAL_C_INCLUDES += \
    $(TARGET_OUT_INTERMEDIATES)/KERNEL_OBJ/usr/include/
 LOCAL_SRC_FILES := ion.c ion_test.c
 LOCAL_MODULE := iontest
 LOCAL_MODULE_TAGS := optional tests

[61 more replies]

closed 2016-01-12 by koquantam