koquantam/android_device_samsung_vivalto3gvn #5: WiFi not work.

ngoquang2708, 2016-01-08:
I open this issue so that all WiFi issues goes here.

koquantam, 2016-01-08:
Wifi is pretty fast compared with stock

closed 2016-01-08 by ngoquang2708

ngoquang2708, 2016-01-08:
Please commit your change to in wpa config.

ngoquang2708, 2016-02-01:
I got this in logcat
'''
W/wpa_supplicant(  613): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant(  613): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant(  613): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant(  613): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant(  613): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant(  613): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant(  613): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant(  613): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant(  613): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant(  613): p2p-dev-wlan0: Failed to initiate AP scan
'''

Do you have it?

koquantam, 2016-02-01:
Stock WPA_supplicant does have it