koquantam/android_device_samsung_vivalto3gvn #6: Phone call not work

ngoquang2708, 2016-01-08:
All ril related issues goes here.

koquantam, 2016-01-08:
If call never ends, change SELinux mode to Permissive

koquantam, 2016-01-08:
When executing rild, i got this
a1@desktop:~$ adb shell
root@vivalto3gvn:/ # rild
255|root@vivalto3gvn:/ # rild -l /system/lib/libreference-ril_sp.so -m w -n 0
rm failed for /data/misc/radio/dlnk, Permission denied
rm failed for /data/misc/radio/dlnk1, Permission denied
rm failed for /data/misc/radio/kmem, Permission denied
rm failed for /data/misc/radio/kmem2, Permission denied
rm failed for /data/misc/radio/ahrh, Permission denied
rm failed for /data/misc/radio/ahrh2, Permission denied
rm failed for /data/misc/radio/mgzc, Permission denied
rm failed for /data/misc/radio/dakl, Permission denied
Opening AT interface. retrying...: No such file or directory
libprotobuf ERROR external/protobuf/src/google/protobuf/message_lite.cc:123] Can't parse message of type "plmn.PLMN_List" because it is missing required fields: version
255|root@vivalto3gvn:/ #

ngoquang2708, 2016-01-08:
modemd and nvitem binary interests me.

koquantam, 2016-01-08:
Got a fix, add nvitemd blob to /system/bin

ngoquang2708, 2016-01-08:
Mobile is managed by ext_*.sh scripts in /system/bin. It seems that those scripts is executed by modemd, I guessed.

koquantam, 2016-01-08:
I think this rom is ready to be release
Most of the important parts working now

koquantam, 2016-01-08:
Maybe, i'll try it
Downloading Asphalt 8 :)

ngoquang2708, 2016-01-08:
Can you successfully make s phone call. Me just hanged.

koquantam, 2016-01-08:
I can receive calls and answer it but If calling anyone, it never ends (even changed SELinux to Permissive)

ngoquang2708, 2016-01-08:
Me, too.

koquantam, 2016-01-08:
The rom seems doesn't too smooth
Where're those "shadows" from? hwcomposer?

ngoquang2708, 2016-01-08:
Hwcomposer lacks some symbols from libbinder (see logcat). So I think it is
hwc problem. It is good with the source code instead of blob.

ngoquang2708, 2016-01-08:
About the hang when you make a call, I think it related to audio recoding process. Sound recorder just hanged like phone call too. We needs more research on this subject.

koquantam, 2016-01-08:
Can't get the dual sim to work. Copied ext_*.sh to /system/bin doesn't making any sense

ngoquang2708, 2016-01-08:
So no mobile data?

koquantam, 2016-01-08:
Yes, SIM can't be registered too

koquantam, 2016-01-08:
Hey, i got the RIL source
checkout this:https://github.com/corphish/android_device_samsung_mint/blob/cm-10.1/ril/SamsungSPRDRIL.java

koquantam, 2016-01-08:
Our device has same SoC compared with Galaxy Star
That device has completed CM11, so we can use this tree as source too

ngoquang2708, 2016-01-08:
More works will get done :)

ngoquang2708, 2016-01-10:
Stock init.rc commented out that rild service because it defines that ril service in init.vivalto3gvn_base.rc. So I commented out rild service init.rc of CM for init.vivalto3gvn_base.rc to take it part. Now RIL work for only one sim. Need to set some prop in build.prop. I am working on it.

koquantam, 2016-01-10:
Yes, i know
Setprop does nothing now (only let it gets the IMEI)

ngoquang2708, 2016-01-10:
That a start. We need to know the way android manage rild and phoneserver daemon. In our case, I think modemd do that dirty job.

koquantam, 2016-01-10:
modemd is included in init.scx15.rc

ngoquang2708, 2016-01-10:
yes, and nvitem do some interest thing to make rild work but I dont know :)

ngoquang2708, 2016-01-10:
I am playing ninja fruit, It is pretty fast.

koquantam, 2016-01-10:
LOL, that's Fruit Ninja man
Asphalt 8 acting like that too :) (even in high graphics, it doesn't FC like stock)

ngoquang2708, 2016-01-10:
OMG, you play Asphalt 8 on this phone :O

koquantam, 2016-01-10:
Yeah, i use GL to SD to put data on SD card

koquantam, 2016-01-10:
hey, i got RIL fully working now

ngoquang2708, 2016-01-10:
One sim?

koquantam, 2016-01-10:
Nope, dual sim..... but it still doesn't register SIM card

ngoquang2708, 2016-01-10:
Wired

koquantam, 2016-01-10:
Alright, will post screenshots

ngoquang2708, 2016-01-10:
how to take screenshot in cm?

koquantam, 2016-01-10:
Keep pressing power button+vol down

ngoquang2708, 2016-01-10:
haha, thanks

koquantam, 2016-01-10:
get high score?

ngoquang2708, 2016-01-10:
No, just testing graphic performance

koquantam, 2016-01-10:
There's sth in stock boot.img that makes RIL working (i flashed it)

koquantam, 2016-01-10:
Nah, still can't make calls

ngoquang2708, 2016-01-10:
You can not make because audio recording problem, not RIL problem. Did you try using Recording App, it cannot record anything and just hang.

koquantam, 2016-01-10:
Yeah, but that's not what i'm gonna say
What i want to say is call nevers ends, it hangs on call ended

koquantam, 2016-01-10:
If you execute the script ril script in init.vivalto3gvn_base.rc, you get Permission denied.

ngoquang2708, 2016-01-10:
I will take a look in logcat to see what is happening.

koquantam, 2016-01-10:
Logcat keep saying this:
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
D/LIGHTS  (  532): file:device/samsung/vivalto3gvn/libs/liblights/lights.c, func:set_light_buttons, on=0
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/        ( 2246): main: slog begin to work. [12]
E/        ( 2246): handle_slog_property: slog.step = 0, slog.reload = 0 [12]
E/        ( 2246): parse_config: cannot find config files! [2]
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/        ( 2249): main: slog begin to work. [12]
E/        ( 2249): handle_slog_property: slog.step = 0, slog.reload = 0 [12]
E/        ( 2249): parse_config: cannot find config files! [2]
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/        ( 2252): main: slog begin to work. [12]
E/        ( 2252): handle_slog_property: slog.step = 0, slog.reload = 0 [12]
E/        ( 2252): parse_config: cannot find config files! [2]
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2182): smd Interface open failed errno is 2 -1
E/SMD     ( 2255): smd_init start 
E/SMD     ( 2255): smd Interface open failed errno is 2 -1
E/SMD     ( 2255): smd Interface open failed errno is 2 -1
E/        ( 2256): main: slog begin to work. [12]
E/        ( 2256): handle_slog_property: slog.step = 0, slog.reload = 0 [12]
E/        ( 2256): parse_config: cannot find config files! [2]
E/SMD     ( 2255): smd Interface open failed errno is 2 -1
E/SMD     ( 2255): smd Interface open failed errno is 2 -1
E/SMD     ( 2255): smd Interface open failed errno is 2 -1
E/SMD     ( 2255): smd Interface open failed errno is 2 -1
E/SMD     ( 2255): smd Interface open failed errno is 2 -1
E/SMD     ( 2255): smd Interface open failed errno is 2 -1

koquantam, 2016-01-10:
I've added some stock init.rc scripts and that's what i get

koquantam, 2016-01-10:
I think that FirefoxOS tree would help us a lot. It just another android OS i think (look at those init scripts)
And i think Sammy just bring Android to our phones use that Spreadtrum chip by that.

koquantam, 2016-01-12:
Damn, got the build without disabling RIL in BoardConfig.mk
Now everytime i unlock the phone, i get "Unfortunately, com.android.systemui has stopped"

ngoquang2708, 2016-01-12:
That the reason I disable RIL

koquantam, 2016-01-12:
Any ideas about this (trying to register SIM with patched telephony)
D/MSimCarrierText(  639): updateCarrierText: text = 
D/MSimCarrierText(  639): updateCarrierText: text = 
D/MSimCarrierText(  639): updateCarrierText: text = NO SERVICE.
D/MSimCarrierText(  639): updateCarrierText: text =

ngoquang2708, 2016-01-12:
Where you get those patchs?

ngoquang2708, 2016-01-12:
I found something at https://gist.github.com/pawitp/9433442. I fork some repo to track changes so please form my repos to sync our changes.

ngoquang2708, 2016-01-13:
After applying those patches, the phone is able to register to not network (one 2g, one 3g, not 3g both). But still showing no service in quick setting icon.

koquantam, 2016-01-13:
Did dialer working?

ngoquang2708, 2016-01-13:
Eh. No.

ngoquang2708, 2016-01-13:
Oh god it dead again.

koquantam, 2016-01-13:
Actually it still dead
http://imgur.com/fRY93c3

koquantam, 2016-01-13:
It doesn't identify phone number

ngoquang2708, 2016-01-13:
There is a bug in ril class, when enable it, the phone shows the same imei for both SIM.

koquantam, 2016-01-13:
SystemUI FC when you "sleep" the phone.

[61 more replies]

closed 2016-02-14 by koquantam