koquantam/android_device_samsung_vivalto3gvn #7: Bluetooth not work.

ngoquang2708, 2016-01-08:


koquantam, 2016-01-08:
Got a fix: copy bluetooth firmware to /system/bin

koquantam, 2016-01-21:
I see there's bluetooth permission and framework in stock

ngoquang2708, 2016-01-22:
Using just libbt-vendor.so from SS make bluetooth work now!

koquantam, 2016-01-22:
What??? It does?
Mine got Bluetooth.apk FC. Did you included in the ROM the permissions and frameworks?

ngoquang2708, 2016-01-22:
done nothing except copy that file from samsung and put it there. reboot.
done.

ngoquang2708, 2016-01-22:
me just fc like yours before

ngoquang2708, 2016-01-22:
it likely that we have a misconfig  of libbt

koquantam, 2016-01-22:
Oh, yeah! Yay! Now let's close this issue

ngoquang2708, 2016-01-22:
it is good to find out what is Ss way of config libbt.

koquantam, 2016-01-22:
So that the first time we don't have bt_vendor.conf and it FC

ngoquang2708, 2016-01-22:
maybe

koquantam, 2016-01-22:
Well, damn. Installed CM and all of the services are DIED

ngoquang2708, 2016-01-22:
try comment ril class in build.prop

ngoquang2708, 2016-01-22:
Or you may regenerate your vendor dir

ngoquang2708, 2016-01-22:
Or you may make a clean build :)

koquantam, 2016-01-22:
Well, i found sth interesting in our stock :D
http://imgur.com/a/C4NKG

koquantam, 2016-01-22:
That was a hidden feature

koquantam, 2016-01-22:
How did you get BT working? I still get that FC

ngoquang2708, 2016-01-22:
How could u do that?

ngoquang2708, 2016-01-22:
Install the ROM from the link I gave u
 Replace libbt-vendor with stock version.

koquantam, 2016-01-22:
Well, i just placed libart.so, libart-compiler.so, dex2oat, oatdump, core-libart.jar/odex to stock system folder and reboot, WHOA!
But it doesn't usable, because stock is odexed and i'm deodexing it, if not all services are died and ART says missing classes.dex
And BT still FC :D

ngoquang2708, 2016-01-22:
I will build it later. Busy now.

ngoquang2708, 2016-01-22:
It is strange, it seems that cm-11 bluedroid version is a bit outdate!!! It does have some function that samsung libbt-vendor.so required. Cm-12.1 does have those function, but I do not know how to update bluedroid in cm-11.0, lots of differences.

koquantam, 2016-01-22:
Well you have to port it, and it's not easy

ngoquang2708, 2016-01-22:
One more thing, we need libbt-hci.so from stock to make bluetooth to work. Sorry about the missing.

koquantam, 2016-01-22:
That's strange. Our phone doesn't have brcm_patchram_plus but Sammy included BT vendor firmware LOL

ngoquang2708, 2016-01-22:
Where is that ```brcm_patchram_plus```?

koquantam, 2016-01-22:
From old Sammy phones, such as S5830i

closed 2016-01-23 by koquantam