koquantam/android_device_samsung_vivalto3gvn #8: Bluetooth not work.

ngoquang2708, 2016-01-08:


ngoquang2708, 2016-01-18:
When using kernel and all *.ko modules compiled from source on stock rom, it can boot normally but something dont work, likes camera... I have no idea.

koquantam, 2016-01-18:
Yeah, stock kernel causes camera FC
Btw, i'm compiling CM12.1 and got this
--------- beginning of main
I/lowmemorykiller(  115): Using in-kernel low memory killer interface
W/auditd  (  113): type=2000 audit(0.0:1): initialized
I/auditd  (  113): type=1403 audit(0.0:2): policy loaded auid=4294967295 ses=4294967295
W/auditd  (  113): type=1404 audit(0.0:3): enforcing=1 old_enforcing=0 auid=4294967295 ses=4294967295
W/init    (   86): type=1400 audit(0.0:4): avc: denied { write } for name="etc" dev="mmcblk0p18" ino=331 scontext=u:r:init:s0 tcontext=u:object_r:system_file:s0 tclass=dir
W/init    (   86): type=1400 audit(0.0:5): avc: denied { write } for name="etc" dev="mmcblk0p18" ino=331 scontext=u:r:init:s0 tcontext=u:object_r:system_file:s0 tclass=dir
W/init    (   86): type=1400 audit(0.0:6): avc: denied { write } for name="etc" dev="mmcblk0p18" ino=331 scontext=u:r:init:s0 tcontext=u:object_r:system_file:s0 tclass=dir
W/init    (   86): type=1400 audit(0.0:7): avc: denied { write } for name="etc" dev="mmcblk0p18" ino=331 scontext=u:r:init:s0 tcontext=u:object_r:system_file:s0 tclass=dir
W/init    (    1): type=1400 audit(0.0:8): avc: denied { setattr } for name="su" dev="mmcblk0p18" ino=1780 scontext=u:r:init:s0 tcontext=u:object_r:su_exec:s0 tclass=file
W/init    (    1): type=1400 audit(0.0:9): avc: denied { write } for name="etc" dev="mmcblk0p18" ino=331 scontext=u:r:init:s0 tcontext=u:object_r:system_file:s0 tclass=dir
W/init    (    1): type=1400 audit(0.0:10): avc: denied { sys_module } for capability=16 scontext=u:r:init:s0 tcontext=u:r:init:s0 tclass=capability
W/init    (    1): type=1400 audit(0.0:11): avc: denied { setattr } for name="mmcblk0" dev="tmpfs" ino=3687 scontext=u:r:init:s0 tcontext=u:object_r:block_device:s0 tclass=blk_file
W/init    (    1): type=1400 audit(0.0:12): avc: denied { setattr } for name="mmcblk0" dev="tmpfs" ino=3687 scontext=u:r:init:s0 tcontext=u:object_r:block_device:s0 tclass=blk_file
W/init    (    1): type=1400 audit(0.0:13): avc: denied { getattr } for path="/dev/mem" dev="tmpfs" ino=3472 scontext=u:r:init:s0 tcontext=u:object_r:kmem_device:s0 tclass=chr_file
I/ServiceManager(  116): service 'media.audio_flinger' died
I/ServiceManager(  116): service 'media.audio_flinger' died

ngoquang2708, 2016-01-18:
There is something to do with selinux enforcing on boot. How to compile kernel with selinux set to permissive?

koquantam, 2016-01-18:
add androidboot.selinux=permissive to BOARD_KERNEL_CMDLINE

ngoquang2708, 2016-01-18:
Did you give it a try?

koquantam, 2016-01-18:
Oh, i forgot. Add it with space, not comma

koquantam, 2016-01-18:
Yes, i'm going to recompile it

ngoquang2708, 2016-01-18:
My plans today was compile cm-12.1, you do, too :)

koquantam, 2016-01-18:
Yeah, i've just started compiling it yesterday. Got 80GB HDD eaten after 8hrs :D

ngoquang2708, 2016-01-18:
Hey, did you try compiled kernel and *.ko modules ON STOCK rom? I tried, and it gives differences result than stock kernel. Why is that?

koquantam, 2016-01-18:
IDK. Camera failed in stock too. Maybe Sammy has done sth :)

ngoquang2708, 2016-01-18:
My 500GiB now left only 50GiB, OMG.

ngoquang2708, 2016-01-18:
Plan for a 1TiB :)

koquantam, 2016-01-18:
If you use ccache, it's even bigger, but for faster builds (since ccache'ed files are copied when compiling)

ngoquang2708, 2016-01-18:
Yeah, that ccache eat my HDD

koquantam, 2016-01-18:
Well that damn SELinux doesn't change to Permissive on boot

ngoquang2708, 2016-01-18:
No idea

ngoquang2708, 2016-01-18:
Compilation stops at: 

target Java: wifi-service (/home/quang/android/cm-12.1/out/target/common/obj/JAVA_LIBRARIES/wifi-service_intermediates/classes)
frameworks/opt/net/wifi/service/java/com/android/server/wifi/WifiController.java:137: error: cannot find symbol
            R.integer.def_wifi_idle_ms);
                     ^
  symbol:   variable def_wifi_idle_ms
  location: class integer
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
1 error

koquantam, 2016-01-18:
Check them up, there's sth that the screen  has passed

koquantam, 2016-01-18:
Wait.... Is that error has red color?

ngoquang2708, 2016-01-18:
Yes, It does.

koquantam, 2016-01-18:
Then you need to check that file in CyanogenMod repo, maybe that's incorrect during source download

ngoquang2708, 2016-01-18:
Found this:

static bool selinux_is_enforcing(void)
{
#ifdef ALLOW_DISABLE_SELINUX
    char tmp[PROP_VALUE_MAX];

    if (property_get("ro.boot.selinux", tmp) == 0) {
        /* Property is not set.  Assume enforcing */
        return true;
    }

    if (strcmp(tmp, "permissive") == 0) {
        /* SELinux is in the kernel, but we've been told to go into permissive mode */
        return false;
    }

    if (strcmp(tmp, "enforcing") != 0) {
        ERROR("SELinux: Unknown value of ro.boot.selinux. Got: \"%s\". Assuming enforcing.\n", tmp);
    }

#endif
    return true;
}

So, add a build prop of ro.boot.selinux=permissive to disable enforcing selinux :), delete old obj/EXECUTABLES/init_...... to have a fresh build.

ngoquang2708, 2016-01-18:
OK, I ll check it out.

koquantam, 2016-01-18:
Still doesn't work
Maybe that's not a problem. Now is the dead media.audio_flinger. The phone stuck on Sammy logo

ngoquang2708, 2016-01-18:
Can I see your logcat?

koquantam, 2016-01-18:
After that, just only "service 'media.audio_flinger' died" :
a1@desktop:~$ adb logcat
--------- beginning of main
I/lowmemorykiller(    0): Using in-kernel low memory killer interface
W/auditd  (  115): type=2000 audit(0.0:1): initialized
I/auditd  (  115): type=1403 audit(0.0:2): policy loaded auid=4294967295 ses=4294967295
W/auditd  (  115): type=1404 audit(0.0:3): enforcing=1 old_enforcing=0 auid=4294967295 ses=4294967295
W/init    (   86): type=1400 audit(0.0:4): avc: denied { write } for name="etc" dev="mmcblk0p18" ino=331 scontext=u:r:init:s0 tcontext=u:object_r:system_file:s0 tclass=dir
W/init    (   86): type=1400 audit(0.0:5): avc: denied { write } for name="etc" dev="mmcblk0p18" ino=331 scontext=u:r:init:s0 tcontext=u:object_r:system_file:s0 tclass=dir
W/init    (   86): type=1400 audit(0.0:6): avc: denied { write } for name="etc" dev="mmcblk0p18" ino=331 scontext=u:r:init:s0 tcontext=u:object_r:system_file:s0 tclass=dir
W/init    (   86): type=1400 audit(0.0:7): avc: denied { write } for name="etc" dev="mmcblk0p18" ino=331 scontext=u:r:init:s0 tcontext=u:object_r:system_file:s0 tclass=dir
W/init    (    1): type=1400 audit(0.0:8): avc: denied { setattr } for name="su" dev="mmcblk0p18" ino=1779 scontext=u:r:init:s0 tcontext=u:object_r:su_exec:s0 tclass=file
W/init    (    1): type=1400 audit(0.0:9): avc: denied { write } for name="etc" dev="mmcblk0p18" ino=331 scontext=u:r:init:s0 tcontext=u:object_r:system_file:s0 tclass=dir
W/init    (    1): type=1400 audit(0.0:10): avc: denied { getattr } for path="/data/local/tmp/slog/slog_sock" dev="mmcblk0p20" ino=47431 scontext=u:r:init:s0 tcontext=u:object_r:shell_data_file:s0 tclass=sock_file
D/NVITEM  (  187): /system/bin/nvitemd /dev/spipe_w1 /system/etc/nvitem_w.cfg FALSE
D/NVITEM  (  187): /system/bin/nvitemd /dev/spipe_w1 /system/etc/nvitem_w.cfg FALSE
D/NVITEM  (  187): is_cali_mode 0 
D/NVITEM  (  187): NVITEM: ramDisk_Init enter
D/NVITEM  (  187): NVITEM:	partId	originImage		backupImage		length
D/NVITEM  (  187): NVITEM:	0x       1	/dev/block/platform/sprd-sdhci.3/by-name/FIXNV1	/dev/block/platform/sprd-sdhci.3/by-name/FIXNV2	0x   40000
D/NVITEM  (  187): NVITEM:	0x       2	/dev/block/platform/sprd-sdhci.3/by-name/RUNTIMENV1	/dev/block/platform/sprd-sdhci.3/by-name/RUNTIMENV2	0x   60000
D/NVITEM  (  187): NVITEM get config finished! 
D/NVITEM  (  187): ramDisk_Read enter
D/NVITEM  (  187): NVITEM partId1:/dev/block/platform/sprd-sdhci.3/by-name/FIXNV2 read success!
D/NVITEM  (  187): ramDisk_Read enter
D/NVITEM  (  187): NVITEM partId2:/dev/block/platform/sprd-sdhci.3/by-name/RUNTIMENV1 read success!
D/NVITEM  (  187): NVITEM:channel open
D/NVITEM  (  187): NVITEM:_getPacket
D/gpsd    (  179): WakeLock(Release,GPSD)
D/gpsd    (  179): WakeLock(Release,SUPL)
D/ATDistributer(  179): [get_agps_cp_interface]: is called
D/ATDistributer(  179): [agps_c_plane_stack_init]: Entry point
D/ATDistributer(  179): open /dev/stty_w21 --> atc0
D/ATDistributer(  179): /dev/stty_w21 is tty device, force it to be raw data mode
D/ATDistributer(  179): [agps_c_plane_stack_init]: Exit point. rc = 0
I/gpsd_d  (  179): GNSS: 206888
I/ServiceManager(  118): service 'media.audio_flinger' died
I/ServiceManager(  118): service 'media.audio_flinger' died
I/ServiceManager(  118): service 'media.audio_flinger' died
I/ServiceManager(  118): service 'media.audio_flinger' died
I/ServiceManager(  118): service 'media.audio_flinger' died
I/ServiceManager(  118): service 'media.audio_flinger' died
I/ServiceManager(  118): service 'media.audio_flinger' died

ngoquang2708, 2016-01-18:
It seems init cannot access files on system partition due to selinux

koquantam, 2016-01-18:
If then, why those later services could run?

koquantam, 2016-01-18:
I think our problem is that dead service

ngoquang2708, 2016-01-18:
W/init ( 86): type=1400 audit(0.0:7): avc: denied { write } for name="etc" dev="mmcblk0p18" ino=331 scontext=u:r:init:s0 tcontext=u:object_r:system_file:s0 tclass=dir

ngoquang2708, 2016-01-18:
Need a permissive kernel right now, bro

koquantam, 2016-01-18:
Hmm....doing it

ngoquang2708, 2016-01-18:
My error is due to out of sync :)

koquantam, 2016-01-18:
Dat damn SELinux needs to be disabled by executing in init scripts

koquantam, 2016-01-18:
Nah, still exist.

ngoquang2708, 2016-01-18:
init binary set it my bro, see the code above

koquantam, 2016-01-18:
Yes, i know, but G313HU done that and success

ngoquang2708, 2016-01-18:
I will try set ro.boot.selinux=permissive and report the result

koquantam, 2016-01-18:
No need. I already tried, and getenforce still gets Enforcing

ngoquang2708, 2016-01-18:
Confirmed.

koquantam, 2016-01-18:
Going to do a clean build with commented-out hardware things to see any differences
Dat takes 5 of 6GB zRAM (LOOOOL)

ngoquang2708, 2016-01-18:
You have 16G ram?

koquantam, 2016-01-18:
Nope, only 4GB man. I don't have that much $$ :D
I set zRAM equals to 3/2 capacity of my RAM, so compiling does not gets "out-of-resource" error.

ngoquang2708, 2016-01-18:
How to set that such high value, I am install zram-config and it automatically set my zram to about 3.6G (nearly half of my ram)

koquantam, 2016-01-18:
Wut??? You have 8GB. Well, idk why linux reserves RAM. Maybe for some reasons
To set zRAM value, do
sudo nano /etc/init/zram-config.conf
find mem= , after totalmem, multiple it as you want (3/2 is a good value, but don't set too high, it's wasted zRAM)

ngoquang2708, 2016-01-18:
Thank you. Sometime, linking libwebviewchromium.so eat my ram up to 7G, just think that 8G is not enough, but 8G is my laptop limit an I cannot effort for another laptop to get 16G :)

koquantam, 2016-01-18:
Your lap is quad-core right?
If so, that RAM usage isn't fit your build

ngoquang2708, 2016-01-18:
What do you mean it isn't fit my build?

ngoquang2708, 2016-01-18:
It processor is an i3 so not really a quad-core

ngoquang2708, 2016-01-18:
And I think that laptop cpu performance is far behind desktop one

koquantam, 2016-01-18:
I mean a quad-core computer must have 16GB RAM to build fast and prevent out-of-memory situation. 
You should upgrade if you can :D
An i3 has dual-core, but Intel has Hyper-Threading in their Core i series, so just know that you have a quad-core computer (the real thing that you have 2 physical cores and 2 logical cores), so build gets faster

ngoquang2708, 2016-01-18:
I understand it now. But 8G is max out of my laptop spec anyway.

koquantam, 2016-01-18:
Yeah, so zRAM is an alternative way to increase your RAM :D

koquantam, 2016-01-18:
Even it's a swap device

ngoquang2708, 2016-01-18:
That is the reason why android use it for low ram device.

ngoquang2708, 2016-01-18:
Trying disabled selinux in kernel config. Hope that the kernel will work without selinux.

koquantam, 2016-01-19:
Have you tried it?

ngoquang2708, 2016-01-19:
Tried. It even cannot mount system.

koquantam, 2016-01-19:
Yeah, of course, because CM12.1 needs SELinux

[97 more replies]

closed 2016-01-31 by koquantam