koquantam/android_device_samsung_vivalto3gvn #9: Audio not work.

ngoquang2708, 2016-01-08:
May be this issue related to phone call not work.

ngoquang2708, 2016-01-12:
I think this issue relates to permissions. Because we can receive incoming call, that mean that the audio recording is working.

ngoquang2708, 2016-01-12:
I will try to look into stock init.rc to find out what permissions is difference from CM one.

koquantam, 2016-01-12:
Of course, it's answerable

ngoquang2708, 2016-01-12:
Int stock init.rc, mediaserver service have system group, in CM, it does not. When add system group to mediaserver in CM init.rc, mediaserver failed to detected the sound card! It is supported to not going that way, because it has more privilege than before. So why adding system group make it fail?

koquantam, 2016-01-12:
Our stock is completely broken. It just does what it wants, like RIL. Our sound works perfectly so i think there's no need to set mediaserver at system group.

ngoquang2708, 2016-01-12:
But when recording audio, logcat split out this thing:

I/MediaFocusControl(  512):  AudioFocus  requestAudioFocus() from android.media.AudioManager@425f2990com.android.soundrecorder.SoundRecorder$4@425e12c8
E/Recorder( 2141): audiosourcetype 1
D/AudioPolicyManagerBase( 1791): getInput() inputSource 1, samplingRate 8000, format 1, channelMask 10, acoustics 0
E/AudioRecord( 1791): Could not get audio input for record source 1
D/AudioSource( 1791): Start autoramp from 500000
E/StagefrightRecorder( 1791): audio source is not initialized
E/MediaRecorder( 2141): start failed: -2147483648

ngoquang2708, 2016-01-12:
When change CM init.rc to stock init.rc, sound work! (and mediaserver has system group, of couse)

koquantam, 2016-01-12:
Problem with the libstagefright:
E/StagefrightRecorder( 1791): audio source is not initialized

ngoquang2708, 2016-01-28:
I have just found a way to make audio rec to work. I will do some check to ensure that it works in our current code base.

koquantam, 2016-01-28:
OK

ngoquang2708, 2016-01-28:
What the f*ck is this
```
I/AudioPolicyService(  131): Loaded audio policy from Samsung hawaii audio policy HAL (audio_policy)
```

koquantam, 2016-01-28:
Hmm... it seems that comes from the property from build.prop
It uses hawaii's policy (hawaii is G313HN/HU board)

ngoquang2708, 2016-01-28:
This issue is now resolved by PR #26

ngoquang2708, 2016-01-29:
After make this to work, I confirm that this issue is not relate to phone call not working issue. Sorry for making many assumtions before.

koquantam, 2016-01-29:
Everybody got their false,man
Dat phone calls refers to RIL, which requires many things to work

ngoquang2708, 2016-01-29:
It seems mint's guys encountered this RIL problem too. I see they are working hard on it.

koquantam, 2016-01-29:
Yeah, mint is a Spreadtrum phone too
He said RIL is working on hardware level (IMEI,signals,etc), but needs to be configured in Sammy/SPRD's way to get it fully working

ngoquang2708, 2016-01-29:
Waiting for his works :D

koquantam, 2016-01-30:
Hmm... i got bootloop when commit this change

ngoquang2708, 2016-01-30:
What change?

koquantam, 2016-01-30:
Audio policy

ngoquang2708, 2016-01-30:
Try a clean build. I encountered it too, but it works after a clean build.

ngoquang2708, 2016-01-30:
May be something doesn't recompile with new config.

koquantam, 2016-01-30:
Damn, 2hrs more :(
Anyway, i've not tested yet because of bootloop, but try disabling CONFIG_SENSOR_OUTPUT_YUV to see if camera works

ngoquang2708, 2016-01-30:
Don't forget to regenerate vendor folder

koquantam, 2016-01-30:
Yes, already

ngoquang2708, 2016-01-30:
I will try it when I have feet time :D

koquantam, 2016-01-30:
It's kernel defconfig, so it's pretty fast to compile

ngoquang2708, 2016-01-30:
Okay

koquantam, 2016-01-30:
Gotta get some sleeps :D

ngoquang2708, 2016-01-30:
It doesn't work.

koquantam, 2016-01-30:
Confirmed

ngoquang2708, 2016-01-30:
Sprd_dcam have less source code than mint kernel one

koquantam, 2016-01-30:
IDK, just digging it :D
But mint is 2mp and ours is 3mp

koquantam, 2016-01-30:
Damn, it crashes mediaserver with stock blob
How did you get it works?

ngoquang2708, 2016-01-30:
I committed all my works, try compare all my repo with yours.

koquantam, 2016-01-30:
Nothing conflicts, but why it still crash?

ngoquang2708, 2016-01-30:
I have no idea

ngoquang2708, 2016-01-30:
Logcat would help.

koquantam, 2016-01-30:
Well, accidentally deleted logfile
But i remember it gives me a lot of backtraces, included audio_policy.scx15.so

ngoquang2708, 2016-01-30:
Very easily to regenerate it.

koquantam, 2016-01-30:
OK, log here: https://pastee.org/z95r8

ngoquang2708, 2016-01-30:
I have no idea. It looks like mine before.

ngoquang2708, 2016-01-30:
I make a clean build, and audio recoding is still working, why don't yours?

koquantam, 2016-01-30:
Idk, typo or sth make it derp?

ngoquang2708, 2016-01-31:
Does it working now?

koquantam, 2016-01-31:
Still dead

koquantam, 2016-01-31:
Sound is dead again

ngoquang2708, 2016-02-02:
Now try remove audio_policy.scx15.so to see what happen.

koquantam, 2016-02-02:
I already tried, missing -DMR0_AUDIO_BLOB in BoardConfig.mk makes audio dead
Just wondering why stock blob works with you, not me

ngoquang2708, 2016-02-02:
I think removing MR0_AUDIO_BLOB make audio recording work with default audio_policy.

koquantam, 2016-02-02:
Then i reports it doesn't work :D
Even @corphish encounters it

ngoquang2708, 2016-02-02:
It is weird!

ngoquang2708, 2016-02-02:
I have some ideas. Report later.

ngoquang2708, 2016-02-02:
There is some SPRD_HARDWARE that I set in system/include/system/audio.h but I cannot send PR to you. See my android_system_core  commits.

koquantam, 2016-02-02:
Lol

ngoquang2708, 2016-02-03:
Great news, audio recording dead again, doing clean build :D

ngoquang2708, 2016-02-03:
This time maybe build faster because I am a noob using ```make bacon``` instead of ```make -j4 bacon``` to use all CPU cores.

ngoquang2708, 2016-02-03:
And one plus point, I use pre-built ```webviewchromium``` now. Check my repo for cherry-pick.

koquantam, 2016-02-03:
Well, don't use make
If you make userdebug build, better use "brunch vivalto3gvn" ( it means lunch cm_vivalto3gvn-userdebug && mka bacon(similar to make but use SCHED_BATCH to ultilitizes CPU cores to make sure it's building at maximum performance))

[17 more replies]

closed 2016-02-06 by koquantam