koquantam/android_device_samsung_vivalto3gvn #24: Wifi tethering

koquantam, 2016-01-23:
Our libhardware_legacy doesn't have wifi p2p support, so it doesn't work

ngoquang2708, 2016-01-23:
I guess CM11 is old compare to A4.4

ngoquang2708, 2016-01-23:
did u try bluetooth telthering

koquantam, 2016-01-23:
There's no bluetooth tethering

ngoquang2708, 2016-01-23:
stock have it

koquantam, 2016-01-23:
Yes, i know, but for our CM11 it doesn't have

ngoquang2708, 2016-01-23:
so sad

koquantam, 2016-01-23:
Hmm... maybe missing files, not due to libhardware_legacy

ngoquang2708, 2016-01-23:
when adding wifi p2p permissions xml. wifi is dead because wpa_supplicant failed to start with p2p support

koquantam, 2016-01-31:
Bro, after wifi commit, you can confirm wifi tethering is fixed: http://imgur.com/a/2FJWw
And you're right, CM has BT tethering

closed 2016-01-31 by koquantam

ngoquang2708, 2016-01-31:
USB tethering to go

koquantam, 2016-01-31:
OK man

ngoquang2708, 2016-01-31:
I cannot use wifi tethering!

ngoquang2708, 2016-01-31:
It shows a bunch of
```
W/wpa_supplicant( 3820): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant( 3820): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant( 3820): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant( 3820): p2p-dev-wlan0: Failed to initiate AP scan
W/wpa_supplicant( 3820): p2p-dev-wlan0: Failed to initiate AP scan
```

koquantam, 2016-01-31:
What?
It's still usable with me

koquantam, 2016-01-31:
Well, there's no such log like that with me
Have you updated init.wifi.rc?

ngoquang2708, 2016-01-31:
Yes

koquantam, 2016-01-31:
OK, i'm out of ideas then

ngoquang2708, 2016-01-31:
At least bt tethering is working so I just left it here.

koquantam, 2016-01-31:
Well you've tested BT tethering right?
I don't have a BT tethering compatible device so can't test it now

ngoquang2708, 2016-01-31:
My laptop have bluetooth. Connected and it can ping google :D

koquantam, 2016-01-31:
At least VT should costs us about 500d for each pings