koquantam/android_device_samsung_vivalto3gvn #33: H/W AVC encoding produces wrong image color while using it in video recoding

ngoquang2708, 2016-01-31:


koquantam, 2016-01-31:
How do you record videos?
I'm using cyanogen_vivalto3g_defconfig now and it just freezes the phone, also no logcat when freezes

ngoquang2708, 2016-01-31:
I think you should make audio recording to work first.

ngoquang2708, 2016-01-31:
Try to pull my latest update from android_system_core to see if it work.

koquantam, 2016-01-31:
Well that doesn't too critical, so i'm gonna fix this first

ngoquang2708, 2016-01-31:
You know, it records audio with video, then...

koquantam, 2016-01-31:
But mic seems to work fine :D

ngoquang2708, 2016-01-31:
But you said that you cannot record audio :D, then...

ngoquang2708, 2016-01-31:
Try delete KERNEL_OBJ and rebuild the kernel to see if it helps. I tried it before and it gave me success.

ngoquang2708, 2016-01-31:
There is no option to set encoding codec in stock camera!!

koquantam, 2016-01-31:
Stock is broken, you know

ngoquang2708, 2016-01-31:
Maybe they can not use avc codec to encode video so they remove those option :D

ngoquang2708, 2016-02-11:
With new omx codecs, seems this issue is not exists anymore, can you check it?

koquantam, 2016-02-11:
It was already worked since you switch to stock libstagefrighthw tho

ngoquang2708, 2016-02-11:
It works, but produces wrong colour. Now it produces the correct one.

koquantam, 2016-02-11:
Confirmed

closed 2016-02-11 by koquantam