koquantam/android_device_samsung_vivalto3gvn #34: Phone suddenly stop working

koquantam, 2016-01-31:
Add it just in case
Needs power_profile.xml editing

ngoquang2708, 2016-01-31:
```
6,3117,381228279,-;PM: suspend entry 2016-01-31 08:28:58.588856281 UTC
4,3118,381228339,-;PM:  before pm_prepare_consol ... 
6,3119,381228406,-;PM: Syncing filesystems...
4,3120,381229780,-;PM:  before pm_notifier_call_chain  ... 
4,3121,381230131,-;*** fw_pm_notify, mode:0x3 done ***
4,3122,381230180,-;*** dmcfreq_pm_notifier_do, event:0x3 ***
4,3123,381230240,-;Freezing user space processes ... 
6,3124,381231006,-;active wake lock sprd-sdhci.0-wakelock
4,3125,381231053,-;
3,3126,381231103,-;Freezing of tasks aborted after 0.000 seconds (60 tasks refusing to freeze, wq_busy=0):
4,3127,381231146,-;
4,3128,381232282,-;sbuf_read wait interrupted!
4,3129,381232438,-;sbuf_read wait interrupted!
4,3130,381232509,-;sbuf_read wait interrupted!
4,3131,381232570,-;sbuf_read wait interrupted!
4,3132,381232633,-;sbuf_read wait interrupted!
4,3133,381232694,-;sbuf_read wait interrupted!
4,3134,381232754,-;sbuf_read wait interrupted!
4,3135,381232814,-;sbuf_read wait interrupted!
4,3136,381232875,-;sbuf_read wait interrupted!
4,3137,381232936,-;sbuf_read wait interrupted!
4,3138,381232997,-;sbuf_read wait interrupted!
4,3139,381233058,-;sbuf_read wait interrupted!
4,3140,381233120,-;sbuf_read wait interrupted!
4,3141,381233280,-;sbuf_read wait interrupted!
4,3142,381233429,-;sbuf_read wait interrupted!
4,3143,381233497,-;sbuf_read wait interrupted!
4,3144,381231185,-;Restarting tasks ... done.
4,3145,381236937,-;*** fw_pm_notify, mode:0x4 done ***
4,3146,381236984,-;*** dmcfreq_pm_notifier_do, event:0x4 ***
6,3147,381237647,-;PM: suspend exit 2016-01-31 08:28:58.598233234 UTC
6,3148,381237711,-;suspend: exit suspend, ret = -16 (2016-01-31 08:28:58.598308468 UTC)
6,3149,381258347,-;PM: suspend entry 2016-01-31 08:28:58.618925328 UTC
4,3150,381258406,-;PM:  before pm_prepare_consol ... 
4,3151,381259425,-;PM:  before pm_notifier_call_chain  ... 
4,3152,381259774,-;*** fw_pm_notify, mode:0x3 done ***
4,3153,381259823,-;*** dmcfreq_pm_notifier_do, event:0x3 ***
4,3154,381259882,-;Freezing user space processes ... 
6,3155,381260635,-;active wake lock wlan_wd_wake
6,3156,381260683,-;active wake lock wlan_rx_wake, time left 50
4,3157,381260727,-;
3,3158,381260776,-;Freezing of tasks aborted after 0.000 seconds (60 tasks refusing to freeze, wq_busy=0):
4,3159,381260818,-;
4,3160,381261927,-;sbuf_read wait interrupted!
4,3161,381262081,-;sbuf_read wait interrupted!
4,3162,381262151,-;sbuf_read wait interrupted!
4,3163,381262212,-;sbuf_read wait interrupted!
4,3164,381262274,-;sbuf_read wait interrupted!
4,3165,381262334,-;sbuf_read wait interrupted!
4,3166,381262394,-;sbuf_read wait interrupted!
4,3167,381262454,-;sbuf_read wait interrupted!
4,3168,381262515,-;sbuf_read wait interrupted!
4,3169,381262574,-;sbuf_read wait interrupted!
4,3170,381262635,-;sbuf_read wait interrupted!
4,3171,381262696,-;sbuf_read wait interrupted!
4,3172,381262757,-;sbuf_read wait interrupted!
4,3173,381262914,-;sbuf_read wait interrupted!
4,3174,381263062,-;sbuf_read wait interrupted!
4,3175,381263129,-;sbuf_read wait interrupted!
4,3176,381260857,-;Restarting tasks ... done.
4,3177,381265397,-;*** fw_pm_notify, mode:0x4 done ***
4,3178,381265444,-;*** dmcfreq_pm_notifier_do, event:0x4 ***
6,3179,381265796,-;PM: suspend exit 2016-01-31 08:28:58.626384828 UTC
6,3180,381265859,-;suspend: exit suspend, ret = -16 (2016-01-31 08:28:58.626456500 UTC)
6,3181,381269136,-;PM: sync done.
6,3182,381269196,-;PM: Syncing filesystems...
6,3183,381273832,-;PM: sync done.
4,3184,381691404,-; RTC ***** interrupt happen
```

koquantam, 2016-01-31:
Where did you get these?
Also, plugging in any power source doesn't has this happens

ngoquang2708, 2016-01-31:
Wifi adb

ngoquang2708, 2016-01-31:
Re-enabling some scx35 defconfig, testing...

ngoquang2708, 2016-01-31:
Would you like to let mali as ko module or built-in moudule?

ngoquang2708, 2016-01-31:
Our boot.img is about 4.7mb, so there is free space.

ngoquang2708, 2016-01-31:
And I am tired of copying mali.ko every time recompiling the kernel.

koquantam, 2016-01-31:
I don't think a built-in module could be done, because init has to insmod it

ngoquang2708, 2016-01-31:
Well, it is built-in so don't need to insmod it!

ngoquang2708, 2016-01-31:
The camera still working after enable previous removed scx35 def! I waiting to see if the phone going to "deep-sleep"

ngoquang2708, 2016-01-31:
And /sys/devices/platform/scxx30-dmcfreq.0/devfreq/scxx30-dmcfreq.0/ondemand/set_freq are back, what the hell, these scx?5 board sharing the same code???

koquantam, 2016-01-31:
They're sprd right?
Look at the c files, there're bunches of #ifdef, so they're the same maybe

ngoquang2708, 2016-01-31:
Anyway, It works now, man.

koquantam, 2016-01-31:
Deep-sleep gone?

ngoquang2708, 2016-01-31:
Yes!

closed 2016-01-31 by koquantam

ngoquang2708, 2016-01-31:
My bad!

koquantam, 2016-01-31:
Anyway, wrong battery capacity causes stats not shown. Please edit power_profile.xml

ngoquang2708, 2016-01-31:
Well
[image]

koquantam, 2016-01-31:
Well, what i meant is only wrong batt capacity causes stats not shown

ngoquang2708, 2016-01-31:
You edited it

ngoquang2708, 2016-01-31:
Could the bat cap dropped and cannot reaches 1500?

koquantam, 2016-01-31:
Check your batt on CPU-Z, does it good or not, you've to replace it