koquantam/android_device_samsung_vivalto3gvn #36: Battery never full, stucks at 99%

ngoquang2708, 2016-01-31:


koquantam, 2016-01-31:
It always full, maybe because of that disabled kernel config before

ngoquang2708, 2016-01-31:
Maybe this is not CM problem, it still showing 99% in stock rom. Needs battery replacement?!

koquantam, 2016-01-31:
Or your charger maybe
What do you use to charge the phone? USB or travelling charger

ngoquang2708, 2016-01-31:
I use travelling charger. I turn the phone off and charge it, now it full.

koquantam, 2016-01-31:
Hah, very funny now
I got it too :D

ngoquang2708, 2016-01-31:
Conclusion: BAD BATTERY

ngoquang2708, 2016-01-31:
So I just leave this issue open unless I buy another new battery!

koquantam, 2016-01-31:
Damn, CPU-Z shows good battery
Maybe because of the battery config? Because i've replaced with a full s5830i battery but still stucks 99

koquantam, 2016-01-31:
OR the kernel

ngoquang2708, 2016-01-31:
I am googling around and see many people complains about this, this is just a common issue of the battery.
Or to be sure, you can switch back to stock an test it.

koquantam, 2016-01-31:
LOL, it's full now after 5 mins of 99 stuck
So that's your batt is dying

ngoquang2708, 2016-01-31:
LOL

closed 2016-02-01 by koquantam