koquantam/android_device_samsung_vivalto3gvn #40: Notification drawer shows "NO SERVICE" instead of operator names

ngoquang2708, 2016-02-02:


koquantam, 2016-02-02:
Well, that's a bug and i've done a workaround with it (already said to you)
Needs init scripts digging to get it working

koquantam, 2016-02-02:
One more thing: Did you registered SIM cards?

ngoquang2708, 2016-02-02:
What you mean "registered"?

koquantam, 2016-02-02:
In Settings >> SIM cards >> Configure SIM cards
You must active first

koquantam, 2016-02-02:
Not by calling 18001090 :D

ngoquang2708, 2016-02-02:
Wait, I don't have ```Configure SIM cards``` option?

ngoquang2708, 2016-02-02:
What will happen if you change those settings?

koquantam, 2016-02-02:
Well, you could use both SIMs for calls and 1st SIM for mobile data
Maybe you only use one SIM so that setting isn't available

ngoquang2708, 2016-02-02:
I have 2 sims, and I cannot do anything with sim 2 except make it register to network by using your workaround.

koquantam, 2016-02-02:
Actually, SIM 2 works on first start after factory reset or clean install and registers both SIMs
RIL2 crashes after reboot

ngoquang2708, 2016-02-06:
This works now.

closed 2016-02-06 by ngoquang2708