koquantam/android_device_samsung_vivalto3gvn #45: Cannot using bluetooth headset for making call

ngoquang2708, 2016-02-02:
When making call, I can switch back and forth between speaker and earpiece but when switch to bluetooth headset, there is no sound. I can listening music through bluetooth headset.

ngoquang2708, 2016-02-02:
Here (logcat)[https://gist.github.com/ngoquang2708/d1b3c80a4719a7992335] when switch to bluetooth headset

koquantam, 2016-02-02:
D/BluetoothManager(  775): isBluetoothAudioConnected: ==> isAudioOn = falsefor headset: 01:20:02:04:C0:23
IDK why it can't connect to BT headset in calls
Is there any other devices you checked?

ngoquang2708, 2016-02-02:
I just have one bluetooth headset. I can connect using a2dp mode to play music on my laptop.

ngoquang2708, 2016-02-02:
I managed to make bluetooth to work using hints from @wakaber repo. You can post this to @corphish to see if this can help him:
```
libbt_vndcfg.txt
```
```
BLUETOOTH_UART_DEVICE_PORT = "/dev/ttyS0"
FW_PATCHFILE_LOCATION = "/vendor/firmware/"
BT_WAKE_VIA_USERIAL_IOCTL = TRUE
LPM_BT_WAKE_POLARITY = 0
LPM_HOST_WAKE_POLARITY = 0
LPM_IDLE_TIMEOUT_MULTIPLE = 5
BTVND_DBG = FALSE
BTHW_DBG = FALSE
VNDUSERIAL_DBG = FALSE
UPIO_DBG = FALSE
FW_PATCH_SETTLEMENT_DELAY_MS = 50
SCO_USE_I2S_INTERFACE = FALSE
SCO_PCM_IF_CLOCK_RATE = 0
PCM_DATA_FMT_FILL_METHOD = 0
PCM_DATA_FMT_FILL_NUM = 0
```

koquantam, 2016-02-02:
Can you try it? I'm a bit busy with RIL now
mint uses different BT firmware, but OK i'll post it

ngoquang2708, 2016-02-02:
When I remove libbt-vendor.so + libbt-hci.so blobs an use CM one, when making call using bluetooth headset, I hear FM signal!

ngoquang2708, 2016-02-02:
OK, just note here. Do not you disturb anymore. Let focus on RIL :D

ngoquang2708, 2016-03-16:
THIS :(

closed 2016-08-04 by ngoquang2708