koquantam/android_device_samsung_vivalto3gvn #46: Not an issue, just CM12.1 discussion

koquantam, 2016-02-05:
Leave this closed

koquantam, 2016-02-05:
It seems that we've to use sth from MTK sources

closed 2016-02-05 by koquantam

ngoquang2708, 2016-02-05:
What are they?

koquantam, 2016-02-05:
IDK yet, but MTK like us, cortex-A7 and it has CM12.1 already

ngoquang2708, 2016-02-05:
OK, interesting. Can you give me some links?

koquantam, 2016-02-05:
This source is for Wiko Rainbow
https://github.com/Al3XKOoL/android_device_wiko_rainbow/tree/cm-12.1

koquantam, 2016-02-07:
YEAHHHHHHHH!!!!!!!!!!!!1
CM12.1 booted

koquantam, 2016-02-07:
Will post screenshots :D

ngoquang2708, 2016-02-07:
Glad to know.

koquantam, 2016-02-07:
Here dude :D:D:D:D:D
[image]

koquantam, 2016-02-07:
It's almost as usable as CM11, but need some config in source to make it fully working as CM11
I forgot the ril-daemon from init.rc so no baseband at that time :D

ngoquang2708, 2016-02-07:
Good job bro.

koquantam, 2016-02-07:
Say "Happy new year" this midnight :D

ngoquang2708, 2016-02-07:
I want to test it bro, link please.

koquantam, 2016-02-07:
OK, wait a little

koquantam, 2016-02-07:
10 mins for first start LOL

koquantam, 2016-02-07:
WTF, mobile data works on SIM 2

ngoquang2708, 2016-02-07:
At least it starts :D

koquantam, 2016-02-07:
I've modified RAMDISK a little, so this build you've to do "start sprd-ril-daemon" and "start sprd-ril-daemon2-3" to start 3 RILs daemon (forgot this :D)

koquantam, 2016-02-07:
https://drive.google.com/file/d/0BzWikRa31-8qS2RBWlI5WmhuSFk/view

koquantam, 2016-02-07:
Missing terminal emulator, FK
You've to do on terminal adb instead :D

ngoquang2708, 2016-02-07:
OK

koquantam, 2016-02-07:
Does it looks well?

ngoquang2708, 2016-02-07:
First start is very slow. But UI is very smooth!

koquantam, 2016-02-07:
Yeah, HWC is enabled, but had to add some config to make it work

koquantam, 2016-02-07:
Alright, i just have pushed working CM12.1 branch

koquantam, 2016-04-03:
Bro, i'm thinking about de-odexing the ROM

koquantam, 2016-04-03:
Since people always use xposed, and after installing, it takes lots of time to recompile the apps btw, so no need for odexing anymore

ngoquang2708, 2016-04-03:
At least, let them boot into the OS quickly to do something.

ngoquang2708, 2016-04-03:
And not everyone use Xposed.

2016-04-03 22:09 GMT+07:00 Quang Ngô <ngoquang2708@gmail.com>: