koquantam/android_device_samsung_vivalto3gvn #54: Phone randomly freeze for no reason

koquantam, 2016-02-08:
Logcat is dead when freezes

koquantam, 2016-02-08:
Confirmed. The overlay commits make the phone freezes immediately

koquantam, 2016-02-09:
Well, @corphish said this is the binder issue. Going to recompile the original bindet source to see it's true or not

ngoquang2708, 2016-02-09:
Do not use libbinder blob

koquantam, 2016-02-09:
I don't use it, and it's freezes like that too

ngoquang2708, 2016-02-09:
Try disable dual sim, I cannot use dual sim at all.

koquantam, 2016-02-09:
Nah, it freezes when running intensive apps, like games.

ngoquang2708, 2016-02-09:
Weird.

koquantam, 2016-02-10:
Could you try playing Subway Surfers to see if it freezes or not?
Reopen it about 3 times

ngoquang2708, 2016-02-10:
Currently I don't have a fast internet connection to download that game. I'll try it later.

koquantam, 2016-02-10:
Any ideas about this:
E/JavaBinder(  534): !!! FAILED BINDER TRANSACTION !!!

koquantam, 2016-02-10:
What the f*ck is this?

> I/Android ( 1552): ######### 1 #########                                                                            
I/Android ( 1552): ######### 11 #########
I/Android ( 1552): ######### 12 #########
I/Android ( 1552): ######### 131 #########
I/Android ( 1552): ######### 132 #########
I/Android ( 1552): ######### 14 #########
I/Android ( 1552): ######### 11 #########
I/Android ( 1552): ######### 12 #########
I/Android ( 1552): ######### 121 #########
I/Android ( 1552): ######### 122 #########
I/Android ( 1552): ######### 123 #########
I/Android ( 1552): ######### 14 #########
I/Android ( 1552): ######### 11 #########
I/Android ( 1552): ######### 12 #########
I/Android ( 1552): ######### 131 #########
I/Android ( 1552): ######### 132 #########
I/Android ( 1552): ######### 14 #########
I/Android ( 1552): ######### 11 #########
I/Android ( 1552): ######### 12 #########
I/Android ( 1552): ######### 121 #########
I/Android ( 1552): ######### 122 #########
I/Android ( 1552): ######### 123 #########
I/Android ( 1552): ######### 14 #########
I/Android ( 1552): ######### 11 #########
I/Android ( 1552): ######### 12 #########
I/Android ( 1552): ######### 131 #########
I/Android ( 1552): ######### 132 #########
I/Android ( 1552): ######### 14 #########
I/Android ( 1552): ######### 11 #########
I/Android ( 1552): ######### 12 #########
I/Android ( 1552): ######### 121 #########
I/Android ( 1552): ######### 122 #########
I/Android ( 1552): ######### 123 #########
I/Android ( 1552): ######### 14 #########
I/Android ( 1552): ######### 11 #########
I/Android ( 1552): ######### 12 #########
I/Android ( 1552): ######### 131 #########
I/Android ( 1552): ######### 132 #########
I/Android ( 1552): ######### 14 #########
I/Android ( 1552): ######### 11 #########
I/Android ( 1552): ######### 12 #########
I/Android ( 1552): ######### 121 #########
I/Android ( 1552): ######### 122 #########
I/Android ( 1552): ######### 123 #########
I/Android ( 1552): ######### 14 #########
I/Android ( 1552): ######### 11 #########
I/Android ( 1552): ######### 12 #########
I/Android ( 1552): ######### 131 #########
I/Android ( 1552): ######### 132 #########
I/Android ( 1552): ######### 14 #########
I/Android ( 1552): ######### 11 #########
I/Android ( 1552): ######### 12 #########
I/Android ( 1552): ######### 131 #########
I/Android ( 1552): ######### 132 #########
I/Android ( 1552): ######### 14 #########
I/Android ( 1552): ######### 11 #########
I/Android ( 1552): ######### 12 #########
I/Android ( 1552): ######### 121 #########
I/ActivityManager(  531): Process com.android.providers.calendar (pid 1243) has died.
V/PanelView(  590): animationTick called with dtms=0; nothing to do (h=112.765076 v=-265.64142)
I/ActivityManager(  531): Process com.android.email (pid 1423) has died.
I/ActivityManager(  531): Process org.cyanogenmod.voiceplus (pid 1520) has died.
I/ActivityManager(  531): Process com.android.gallery3d (pid 1505) has died.
I/ActivityManager(  531): Process com.android.calendar (pid 1321) has died.
I/ActivityManager(  531): Process com.cyanogenmod.account (pid 1590) has died.

koquantam, 2016-02-10:
People confuse a lot about this, so i've to solve it quickly

koquantam, 2016-02-11:
Confirmed. Kernel issue

closed 2016-02-11 by koquantam