koquantam/android_device_samsung_vivalto3gvn #72: Cannot use speaker in call (without audio blob)

ngoquang2708, 2016-02-13:


koquantam, 2016-02-13:
Speaker in calls still usable

ngoquang2708, 2016-02-13:
It is usable with stock blobs, not with our code.

ngoquang2708, 2016-02-16:
I know why FFOS's audio primary module cannot do audio ril commands to switch the audio output to speaker. It is because of our libatchannel failed to connect to ril dameon to send AT commands. Stock libatchannel.so uses libsecril-client.so to connect to ril daemon to be able to send AT commands.
What we need to to here is to find the way to send AT commands to ril daemon and implement it in libatchannel. Using libsecril-client.so is acceptable since it API is [here](https://github.com/CyanogenMod/android_hardware_samsung/blob/cm-12.1/ril/libsecril-client/secril-client.h).

ngoquang2708, 2016-02-16:
Here logcat when stock audio blobs doing switching
```
W/audio_hw_primary(  128): [out_set_parameters], kvpairs: routing=2, devices: 0x1, mode: 2 
W/audio_hw_primary(  128): [out_set_parameters],val=0x2 adev->prev_out_device=0x1,adev->out_device=0x1
W/audio_hw_primary(  128): out_set_parameters want to set devices:0x1 old_mode:2 new_mode:2 call_start:1 val=0x2
W/audio_hw_primary(  128): i2s_pin_mux_sel in type is 0
D/audio_hw_primary(  128): do_cmd_dual Switch incall AT command [AT+SSAM=1][find sril_SendATHandler] 
W/audio_hw_primary(  128): out_set_parameters EXTRAVOLUME
W/audio_hw_primary(  128): at_cmd_extra_volume, enable:0
D/audio_hw_primary(  128): do_cmd_dual Switch incall AT command [AT+SPAUDIOCONFIG=extravgr,0][find sril_SendATHandler] 
```

ngoquang2708, 2016-02-16:
And this is the interesting thing
```
[find sril_SendATHandler]
```

corphish, 2016-02-16:
Probably Sammy modified both of them.
Anyway, you could try changing  user,group of various modem related services in init.*.rc
Like modemd, phoneserver, ril-daemon etc.
Probably see how init is defined in FFOS

koquantam, 2016-02-16:
Looks like both you guys hate stock blobs yeah :D
Btw, i'm still working on CM12.1 to get this usable. Really liked what Sammy does to ION driver. Going to port and test this.
Well, good luck!

ngoquang2708, 2016-02-17:
Do you have any ideas about how to use libsecril-client's InvokeOemRequestHookRaw() function to send AT commands?

ngoquang2708, 2016-02-17:
I got this from stock audio blobs when switching to speaker:
```
D/use-Rlog/RLOG-RIL(  157): [w] onRequest: OEM_HOOK_RAW sState=4
D/use-Rlog/RLOG-RIL(  157): [w] channel1 state: '0' 
D/use-Rlog/RLOG-RIL(  157): [w] get Channel ID '1'
D/use-Rlog/RLOG-AT(  157): [w] Channel1: AT> AT+SSAM=1
D/use-Rlog/RLOG-AT(  157): [w] Channel1: AT< OK
E/use-Rlog/RLOG-RILC(  157): [w] RIL_onRequestComplete: invalid RIL_Token
D/use-Rlog/RLOG-RIL(  157): [w] put Channel ID '1'
D/use-Rlog/RLOG-RIL(  157): [w] onRequest: OEM_HOOK_RAW sState=4
D/use-Rlog/RLOG-RIL(  157): [w] channel1 state: '0' 
D/use-Rlog/RLOG-RIL(  157): [w] get Channel ID '1'
D/use-Rlog/RLOG-AT(  157): [w] Channel1: AT> AT+SPAUDIOCONFIG=extravgr,0
D/use-Rlog/RLOG-AT(  157): [w] Channel1: AT< OK
E/use-Rlog/RLOG-RILC(  157): [w] RIL_onRequestComplete: invalid RIL_Token
D/use-Rlog/RLOG-RIL(  157): [w] put Channel ID '1'
D/use-Rlog/RLOG-RIL(  157): [w] onRequest: OEM_HOOK_RAW sState=4
D/use-Rlog/RLOG-RIL(  157): [w] channel1 state: '0' 
D/use-Rlog/RLOG-RIL(  157): [w] get Channel ID '1'
D/use-Rlog/RLOG-AT(  157): [w] Channel1: AT> AT+VGR=6
D/use-Rlog/RLOG-AT(  157): [w] Channel1: AT< OK
E/use-Rlog/RLOG-RILC(  157): [w] RIL_onRequestComplete: invalid RIL_Token
D/use-Rlog/RLOG-RIL(  157): [w] put Channel ID '1'
D/use-Rlog/RLOG-AT(  161): [w] Channel3: AT< +CSQ: 14,99
D/use-Rlog/RLOG-AT(  161): [w] Channel3: AT< +SPSCSQ: 14,99,1
```

ngoquang2708, 2016-02-17:
Found a hack by using stock libatchannel, but it sendAt function API is a bit different from FFOS one:
```
sendAt(void *unknown, int simId, const char *atCmd, void *unknown2)
```
instead of:
```
sendAt(void *buf, size_t bufLen, int simId, char *atCmd)
```

ngoquang2708, 2016-02-18:
This should be fixed by PR #77

koquantam, 2016-02-18:
OK, i'll do it later. For now i can't

koquantam, 2016-02-21:
Bro, left those sources like audio, OMX, etc into common vendor, since not only our device use them.

closed 2016-02-21 by koquantam

ngoquang2708, 2016-02-21:
But I just config them to work with our device, I do not have time to make them universal.