koquantam/android_device_samsung_vivalto3gvn #83: phoneserver failed to set ril.sim2.present prop

ngoquang2708, 2016-02-27:


ngoquang2708, 2016-02-27:
Why phoneserver have uid of 1000 (system) while in init.scx15.rc:
```
service phoneserver_w /system/bin/phoneserver -m w
    class core
    user root
    group radio system shell
    disabled
    oneshot
```
@koquantam @Y300-0100

Y300-0100, 2016-02-27:
user root
           0
group radio system shell
            1001   1000    2000

it look correct???
what give you the problem?

ngoquang2708, 2016-02-27:
```ps``` shows that ```phoneserver``` is running as ```system```, not ```root``` as in ```init.scx15.rc```

ngoquang2708, 2016-02-27:
And because of being run as ```system``` user, ```phoneserver``` cannot set ```ril.sim2.present``` prop. And we have to hack by manually set it in ```system.prop``` file, that is not so good.

ngoquang2708, 2016-02-27:
mint's dev faced this problem too.
One thing to point out that, int ```system/core/init/property_setvice.c```, line ```74``` ```    { "ril.",             AID_RADIO,    0 },```.

ngoquang2708, 2016-02-27:
@corphish

Y300-0100, 2016-02-27:
As I can remember some Chinese dev named hacker something dsd had to be set in build prop even for cm 10

Y300-0100, 2016-02-27:
https://github.com/HackerOO7
it took me a while.

Y300-0100, 2016-02-27:
did u tried this
persist.multisim.config=dsds
persist.radio.multisim.config=dsds
ro.multi.rild=true

ngoquang2708, 2016-02-27:
What do these prop do?

koquantam, 2016-02-27:
LOL, that's for dual-SIM support only. Our RIL is different, not getting it work easily with normal scripts. Needs to hack the init as @ngoquang2708  said

ngoquang2708, 2016-02-27:
@koquantam I don't know what you mean. Didn't we do support for Dual-SIM?

koquantam, 2016-02-27:
Sorry, I'm answering @Y300-0100's question :/
Btw, @corphish has successfully ported separate ION driver for SPRD phones, at least with mint. It works with CM11 unpatched binder. Hope it works with CM12.1

Y300-0100, 2016-02-27:
I don't know does this will help:

https://github.com/sprd-dev/device_sprd/blob/sprdroid4.4/scx15/init.scx15.rc

ngoquang2708, 2016-02-27:
@koquantam our camera blobs relies on ```libbinder``` for ```MemoryHeapIon``` functions, I do not think what we could make ```MemoryHeapIon``` as a serparate library.

ngoquang2708, 2016-02-27:
@Y300-0100 thanks for pointing it out, our board has ```init.scx15.rc```.

ngoquang2708, 2016-02-27:
What kill me right now is why ```phoneserver``` is declared to run as ```root``` user, but actually running as ```system``` user :(

koquantam, 2016-02-27:
LD_PRELOAD does that
Could you try removing system group from phoneserver?

ngoquang2708, 2016-02-27:
@koquantam I will give it a try.
@Y300-0100 Did you build this ROM successfully? Any errors?

Y300-0100, 2016-02-27:
syncing sources. 50%

ngoquang2708, 2016-02-27:
@Y300-0100 A tips for you, build the kernel first, because some of our HAL modules relies on kernel header to be able to build.

Y300-0100, 2016-02-27:
I noticed that. missing <include/...>

Y300-0100, 2016-02-27:
include/linux etc

ngoquang2708, 2016-02-27:
Yes, build the kernel first and android build system will prepare those headers for us.

koquantam, 2016-02-27:
libmemoryheapion patch:
https://github.com/TeamButter/android_device_samsung_sprd-common/commit/873f22752ce548ae9f7ac615dc62bf92ef3bff3d

koquantam, 2016-02-27:
Sorry for off-topic

Y300-0100, 2016-02-27:
I was managed to build one previously but com.android.phone has stopped was killing me. any ideas on that?

ngoquang2708, 2016-02-27:
May be you did not remove default ```ril-daemon``` service in ```init.rc```, it conflicts with ```ril-daemon``` services in ```init.vivalto3gvn_base.rc```.

koquantam, 2016-02-27:
Don't just rm -rf'ing the system folder. You have to fully install the ROM. Stock blobs sometimes give us headache

Y300-0100, 2016-02-27:
I did full wipe but didn't remove default ril-daemon

koquantam, 2016-02-27:
Default RIL-daemon just gives you no signals, else no errors.

Y300-0100, 2016-02-27:
U mean this:
#service ril-daemon /system/bin/rild
#    class main
#    socket rild stream 660 root radio
#    socket rild-debug stream 660 radio system
#    user root
#    group radio cache inet misc audio log

ngoquang2708, 2016-02-27:
Yes, I commented it out, so no conflicts.

ngoquang2708, 2016-02-27:
OK, some news, I patches ```init```'s ```property_service.c``` to add permission for ```system``` user to be able to set ```ril.``` prop, then I remove ```ril.sim2.present``` in ```build.prop``` and sim2 still work, no deny from dmesg.

Y300-0100, 2016-02-27:
I don't have
 #wcdma rild service
 service ril-daemon /system/bin/rild -l /system/lib/libreference-ril_sp.so -m w -n 0

in my init.kanas3gnfc_base.rc

ngoquang2708, 2016-02-27:
May be they are in init.scx15.rc

ngoquang2708, 2016-02-27:
I have taken a look, it is here ```init.sc8830.rc```

Y300-0100, 2016-02-27:
yes but it's
 service tril-daemon
[init.sc8830.rc.txt](https://github.com/koquantam/android_device_samsung_vivalto3gvn/files/149453/init.sc8830.rc.txt)

ngoquang2708, 2016-02-27:
Of course, they are different board!

koquantam, 2016-02-27:
@Y300-0100
tril-daemon is for TD-SCDMA phones, that only supports 2G mobile data. You can remove it

Y300-0100, 2016-02-27:
@koquantam 
Thanks
I just compare ih24n69 root files. They are same as mine so it must be something else..

ngoquang2708, 2016-02-27:
Using ```meld```?

Y300-0100, 2016-02-27:
@ngoquang2708
Using meld?
Don't know what you mean?

ngoquang2708, 2016-02-27:
```meld``` is tool to compare, merge,... It is worth trying it.

Y300-0100, 2016-02-28:
device/samsung/kanas3gnfcxx/libs/libatchannel/AtChannel.cpp:188:22: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
device/samsung/kanas3gnfcxx/libs/libatchannel/AtChannel.cpp:189:68: error: 'Connect_RILD_Second' was not declared in this scope
make: *** [/home/darkside/android/out/target/product/kanas3gnfcxx/obj/SHARED_LIBRARIES/libatchannel_intermediates/AtChannel.o] Error 1
make: *** Waiting for unfinished jobs....
device/samsung/kanas3gnfcxx/libs/audio/audio_hw.c: In function 'out_write':
device/samsung/kanas3gnfcxx/libs/audio/audio_hw.c:2283:8: warning: 'ret' may be used uninitialized in this function [-Wmaybe-uninitialized]

ngoquang2708, 2016-02-28:
@Y300-0100 You need to add ```COMMON_GLOBAL_CFLAGS += -DSEC_PRODUCT_FEATURE_RIL_CALL_DUALMODE_CDMAGSM``` in your ```BoardCnnfig.mk```. Btw, I don't know if our ```libatchannel``` and ```libsecril-client``` work for you, because different ```rild``` has different method to communicate with its clients.

Y300-0100, 2016-02-28:
@ngoquang2708 
But I'm trying :)
For now I changed this to bypass error:
`int (*connect_rild)(HRilClient client) = i == 0 ? Connect_RILD : Connect_RILD;`

ngoquang2708, 2016-02-28:
Do not do that, add ```COMMON_GLOBAL_CFLAGS += -DSEC_PRODUCT_FEATURE_RIL_CALL_DUALMODE_CDMAGSM``` in ```BoardConfig.mk```, that ```Connect_RILD``` function is definfed in ```libsecril-client```, read it ```Android.mk``` for more details.

ngoquang2708, 2016-02-28:
Fixed as of https://github.com/ngoquang2708/android_system_core/commit/07097e72a70225e270ad6164ad7ce1aed5106276.

closed 2016-02-28 by ngoquang2708