koquantam/android_device_samsung_vivalto3gvn #86: cm-12.1: needs a proper SELinux configuration

ngoquang2708, 2016-03-02:


koquantam, 2016-03-02:
Well, @halfpsych said that we've to configure a lot from the kernel, by adding policies. This would give us a lot of time, and it's a PITA. Do it if you really have time

ngoquang2708, 2016-03-02:
I do not have time, noted here in case I forgot it.

ngoquang2708, 2016-03-02:
Do you know to to use other GCC version. For unknown reason (may be I've updated Ubuntu to 16.04), when I build cm-12.1, it use GCC 4.6.

koquantam, 2016-03-02:
I'm using Linaro for my builds, but it causes a lot segmentation faults
Btw, if you want to use other toolchains, do
1. Go to build/core/combo/TARGET_linux-arm.mk
- Edit "TARGET_GCC_VERSION" , "TARGET_TOOLCHAIN_ROOT" , "TARGET_TOOLS_PREFIX" to match your toolchain
2. Go to build/envsetup.sh
- Go to line 156, edit again to match your toolchain

ngoquang2708, 2016-03-02:
It failed me at ```HOST_*``` things. Any way, thanks.

koquantam, 2016-03-02:
I recommend using Linaro 4.8 for CM12.1, since it optimizes a lot for our ARM instruction set.

koquantam, 2016-03-03:
Seems CM12.1 toolchain works with CM11

ngoquang2708, 2016-03-03:
Tried Linaro 4.7 for cm11, kernel works fine, but crashing surfaceflinger.

koquantam, 2016-03-03:
Set SELinux to Permissive
That work for me. IDK why it affects, it just a kernel implementation

ngoquang2708, 2016-03-03:
Didn't SELinux is already permissive?

koquantam, 2016-03-03:
I thought you changed it to Enforcing
There's a weird thing here: You will see the benchmark result decreases a lot from original GCC toolchain, but performance increase a lot. Very weird here

ngoquang2708, 2016-03-03:
Currently, trying to clean ```ccache``` and doing anything again :(

koquantam, 2016-03-03:
ccache -C will do that

ngoquang2708, 2016-03-03:
Yes, waiting for it to complete, lazy of reconfig it.

ngoquang2708, 2016-03-13:
Move to https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/issues/28.

closed 2016-03-13 by ngoquang2708