koquantam/android_device_samsung_vivalto3gvn #102: 404 Link not found

Grace5921, 2016-11-07:
https://github.com/ngoquang2708/android_external_sepolicy/compare/ngoquang2708:f00429df5685a46aa4f4694dab8f68d6d5645cd0...cm-13.0

This link is not working . by the way i have modifed your tree work on xolo era 4g (sc8830) 

https://github.com/era4g/android_device_xolo_era4g 

but not tested yet can you help me with it (it has not got kernel source but uses 6.0 has default os )

koquantam, 2016-11-07:
That patch is deprecated, so there's no need for it

closed 2016-11-07 by koquantam

Grace5921, 2016-11-07:
ok thanks by the way can you help little bit ?

koquantam, 2016-11-08:
Yes, just ask what you need

Grace5921, 2016-11-08:
I have compiled twrp using your tree here is source https://github.com/era4g/android_device_xolo_era4g/tree/twrp and it's not booting (actually it boots but sucked on logo and i can use adb and all shell commands are working )

here is logcat of twrp:-  http://hastebin.com/raw/rororihale

Thanks for any help .

Grace5921, 2016-11-27:
sorry it's only log

ngoquang2708, 2016-12-01:
@Grace5921 ```TARGET_RECOVERY_FSTAB``` is used by default recovery which has the same syntax as normal fstab so you could use the same fstab for both normal and recovery mode (at least for M, I think). TWRP uses a difference recovery fstab which should be put in ```recovery/root/etc/twrp.fstab```. You could use ```PRODUCT_COPY_FILES``` to copy your current ```recovery.fstab``` to that location. Something like this
```
PRODUCT_COPY_FILES += \
    device/vendor/codename/rootdir/twrp.fstab:recovery/root/etc/twrp.fstab
```

Grace5921, 2016-12-01:
ya i did but recovery not booted so i left working on that device
 any ways thanks